vectorgrp/XCPlite #3: Request for clarification

RudolphRrr, 2021-05-17:
Hello,

I just found this repository and while the whole thing is released under MIT license,
the individual files do not have license information.

For example, what is the supposed status of vxlapi.h?
It clearly states Vector Informatik copyright, no issue there, but what about the license?
Or A2L.c, there is no copyright or license for this file.

RainerZ, 2021-06-14:
Hello,

our understanding is, that the MIT license does not require the license text to be included inside each source file.

The license of vxlapi.h is indeed unclear, as the official download of the library installer states different license conditions. 
We will simplify XL driver library licensing in the future.
I will remove the files here.
Please download the official "XL Driver Library" Setup from www.vector.com.

RudolphRrr, 2021-06-15:
>our understanding is, that the MIT license does not require the license text to be included inside each source file.

And in general I agree.
However, you are mixing files from several sources here, like any normal project would do.
And assessing afterwards which files came under which license is a real pain with no hints in the files and sometimes
even no copyright information.

Sure, you are perfectly fine, even more so since this repository is marked to be from Vector.
But for everyone else this is a copyright minefield.

For example A2L.h and A2L.c.
Lets assume one would like to use these, finds this repository, sees MIT license and copies the files into their own project.
Now someone else is conducting a BlackDuck test a while later and BlackDuck determines there is a potential license conflict as these two files were found in a totally different repository that is under GPL.
There is not a even a hint that A2L.h and A2L.c are under Vector copyright and no license.
Putting a standard header in the file including the license would make things so much easier.

>The license of vxlapi.h is indeed unclear, as the official download of the library installer states different license conditions.

The XL Driver Library is distributed without any license currently, I hope that gets changed soon.

RainerZ, 2021-06-16:
Ok, thanks for your comments.

I just added copyright notes to all source files.

Until the XL-API license conditions are clarified and simplified, please contact Vector for an individual XL Driver Library license agreement.

closed 2021-06-28 by RainerZ