vectorgrp/XCPlite #4: Compile xcpLite under LINUX / Ubuntu 18.04

brenkem, 2021-06-18:
Hello.

I tried to compile the CMake project under my Ubuntu 18.04 system, but failed (see [1]).

What is the correct way to build xcpLite?

Thanks.

[1]:

```
$ git clone https://github.com/vectorgrp/XCPlite.git
$ mkdir build
$ cd build/
$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/brenkem/git/XCPlite/build

$ make
Scanning dependencies of target xcpLite
[  9%] Building CXX object CMakeFiles/xcpLite.dir/main.cpp.o
In file included from /home/brenkem/git/XCPlite/xcpLite.h:1641:0,
                 from /home/brenkem/git/XCPlite/main.h:137,
                 from /home/brenkem/git/XCPlite/main.cpp:15:
/home/brenkem/git/XCPlite/main.cpp: In function ‘int createA2L(const char*)’:
/home/brenkem/git/XCPlite/xcpAppl.h:42:41: error: cast from ‘vuint8* {aka unsigned char*}’ to ‘vuint32 {aka unsigned int}’ loses precision [-fpermissive]
 #define ApplXcpGetAddr(p) ((vuint32)((p)))
                                         ^
/home/brenkem/git/XCPlite/A2L.h:59:114: note: in expansion of macro ‘ApplXcpGetAddr’
 ameterWithLimits(name,comment,unit,min,max) A2lCreateParameterWithLimits_(#name,sizeof(name),ApplXcpGetAddr((vuint8*)&name),comment,unit,min,max)
                                                                                              ^~~~~~~~~~~~~~
/home/brenkem/git/XCPlite/main.cpp:53:5: note: in expansion of macro ‘A2lCreateParameterWithLimits’
     A2lCreateParameterWithLimits(gDebugLevel, "Console output verbosity", "", 0, 4);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/brenkem/git/XCPlite/xcpAppl.h:42:41: error: cast from ‘vuint8* {aka unsigned char*}’ to ‘vuint32 {aka unsigned int}’ loses precision [-fpermissive]
 #define ApplXcpGetAddr(p) ((vuint32)((p)))
                                         ^
/home/brenkem/git/XCPlite/A2L.h:59:114: note: in expansion of macro ‘ApplXcpGetAddr’
 ameterWithLimits(name,comment,unit,min,max) A2lCreateParameterWithLimits_(#name,sizeof(name),ApplXcpGetAddr((vuint8*)&name),comment,unit,min,max)
                                                                                              ^~~~~~~~~~~~~~
/home/brenkem/git/XCPlite/main.cpp:54:5: note: in expansion of macro ‘A2lCreateParameterWithLimits’
     A2lCreateParameterWithLimits(gFlushCycleMs, "DAQ flush cycle time, 0 = off", "", 0, 1000);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/brenkem/git/XCPlite/main.cpp: In function ‘int main(int, char**)’:
/home/brenkem/git/XCPlite/main.cpp:186:49: warning: too many arguments for format [-Wformat-extra-args]
         else if (sscanf_s(argv[i], "-t%c", &c, 1) == 1) {
                                                 ^
CMakeFiles/xcpLite.dir/build.make:62: recipe for target 'CMakeFiles/xcpLite.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/xcpLite.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/xcpLite.dir/all' failed
make[1]: *** [CMakeFiles/xcpLite.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

RainerZ, 2021-06-22:
Hi,

seems to be 64 bit target. The Linux version was tested on a RaspberryPi as ARM32 target variant so far.
I created a new branch Linux64 which I could compile (via vscode and  g++) on Ubuntu 20.04 in 64 bit mode.
Added a quick and dirty handling for 64 bit to 32 bit A2L address conversion, which works ok with the unmodified generated A2L. 
Will probably not work with ELF linker map updater yet. 64 Bit mode ELF has not been verified.
Maybe you want to give this a try.

Regards

brenkem, 2021-06-23:
Thanks a lot. The tool can now also be compiled under 64bit Linux systems, see [1]. Also the demonstration is running after executing xcpLite, see [2].

We can now close the issue.

[1]:
$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/brenkem/git/XCPlite/build

$ make
Scanning dependencies of target xcpLite
[  9%] Building CXX object CMakeFiles/xcpLite.dir/main.cpp.o
[ 18%] Building CXX object CMakeFiles/xcpLite.dir/ecupp.cpp.o
[ 27%] Building C object CMakeFiles/xcpLite.dir/ecu.c.o
[ 36%] Building C object CMakeFiles/xcpLite.dir/udpserver.c.o
[ 45%] Building C object CMakeFiles/xcpLite.dir/udp.c.o
[ 54%] Building C object CMakeFiles/xcpLite.dir/xcpAppl.c.o
[ 63%] Building C object CMakeFiles/xcpLite.dir/xcpLite.c.o
[ 72%] Building C object CMakeFiles/xcpLite.dir/A2L.c.o
[ 81%] Building C object CMakeFiles/xcpLite.dir/xcpSlave.c.o
[ 90%] Building C object CMakeFiles/xcpLite.dir/clock.c.o
[100%] Linking CXX executable xcpLite
[100%] Built target xcpLite

[2]:
 $ ./xcpLite 

XCPlite Demo - ECU Simulator with XCP on Ethernet (UDP)
Vector Informatik GmbH 2021
Build Jun 23 2021 12:01:21

Options:
XCPSIM_ENABLE_A2L_GEN,CLOCK_USE_UTC_TIME_NS

Init clock 
  System realtime clock resolution = 0s,1ns
  +0us:   23.6.2021 12:01:29 +203236657
  +100us: 23.6.2021 12:01:29 +203399308

Init XCP protocol layer
  (Version=1.4, MAXEV=256, MAXCTO=250, MAXDTO=1396, DAQMEM=500000, MAXDAQ=65535, MAXENTRY=65535, MAXENTRYSIZE=255)
  (ENABLE_CHECKSUM,DAQ_CLOCK_64BIT,XCP_ENABLE_A2L_NAME,XCP_ENABLE_A2L_UPLOAD,XCP_ENABLE_DAQ_EVENT_LIST,)
  Listening on port 5555
  Bind sin_family=2, addr=0.0.0.0, port=5555

Event 0: Cyclic unit=4 cycle=200 samplecount=0
Event 1: Task1 unit=4 cycle=200 samplecount=0
Event 2: Task2 unit=4 cycle=200 samplecount=0
Event 3: TaskAct unit=3 cycle=0 samplecount=0

Create A2L XCPlite.A2L
BaseAddr = 55F000000000
  (45 meas, 26 pars, 1 typedefs, 6 components, 3 instances, 13 conversions)

Start C++ demo task (cycle = 2000us, ext event = 3, size = 88 )
Start XCP DAQ thread
Start C demo task (cycle = 2000us, event = 0)
Start XCP server thread

closed 2021-06-23 by brenkem