vectorgrp/XCPlite #6: CMakeLists

yurijgera, 2021-07-17:
CMake project generation fails on master.
udpserver.c was renamed to xcpTl.c

Build still fails due to incomplete type ‘struct dl_phdr_info’
The reason seem to be that cmake build mixes compilers and uses c++ for .cpp and cc for .c files.
The one does seem to define _GNU_SOURCE while the other don't. Uncommenting _GNU_SOURCE in the code produces redefinition warning for the compiler that already defined the _GNU_SOURCE.

Since XCPlite does seem to rely on GNU and the vscode uses GNU g++ compiler for all .c and .cpp files, I suggest enforcing that compiler in the CMakeLists.txt by adding:

set(CMAKE_C_COMPILER "g++")
set(CMAKE_CXX_COMPILER "g++")

RainerZ, 2021-11-13:
Sorry, I left that issue open 
The change has been added

RainerZ, 2021-11-13:
...

closed 2021-11-13 by RainerZ