vectorgrp/XCPlite #8: Provide a manual

sledge33, 2021-11-09:
Hi,
I m new to xcp topic and found your repository. Could you provide a tutorial (pdf) how to get this run, e.g. a raspberry pi as Slave and a Windows system as Master?

RainerZ, 2021-11-13:
Thanks for your suggestion.
We will provide a short documentation how to get this runing on a raspberry pi and the CANape demo version.
This will take a while - currently busy with the CANape 20 release testing.

If you can not wait and you managed to build and run XCPlite, just do this in CANape 19:
Load Project - choose CANape.INI in the XCPlite/CANape folder
Go to Device Configuration/Protocol/Transport Layer Settings and change the ECU IP address to that of your Raspi
Press the Online button in the toolbar
Press the Start/Stop Measurement button in the toolbar
Go to Measurement Configuration to add new measurement signals

sledge33, 2022-08-06:
are there any updates on this topic? documentation / howto manual available?

RainerZ, 2022-08-31:
Sorry, not yet
Some efforts for this are planned during September

closed 2025-07-26 by RainerZ