vectorgrp/XCPlite #10: Compile xcpLite under Ubuntu

ayoubziat, 2021-11-14:
Hello,

I tried to compile the CMake project, but got several errors. 
Could you please fix this? 

I will attach later on the errors.

Thanks

frederich, 2021-12-02:
Which errors occurred, details please?

ayoubziat, 2021-12-02:
Hi, sorry for my late reply.

Following errors occur when compiling the cmake project under Ubuntu /WSL.

```
~/xcpTl.c:421:46: error: invalid conversion from ‘char*’ to ‘uint8_t*’ {aka ‘unsigned char*’} [-fpermissive]
  421 |         n = socketRecv(gXcpTl.MulticastSock, (char*) buffer, (uint16_t)sizeof(buffer));
```

```
/usr/bin/ld: CMakeFiles/xcpLite.dir/xcpLite.c.o: in function `XcpCommand':
~/xcpLite.c:1116: undefined reference to `XcpTlSetClusterId'
/usr/bin/ld: CMakeFiles/xcpLite.dir/xcpLite.c.o: in function `XcpInit':
~/xcpLite.c:1349: undefined reference to `XcpTlSetClusterId'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/xcpLite.dir/build.make:219: xcpLite] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/xcpLite.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

frederich, 2021-12-06:
> Hi, sorry for my late reply.
> 
> Following errors occur when compiling the cmake project under Ubuntu /WSL.
> 
> ```
> ~/xcpTl.c:421:46: error: invalid conversion from ‘char*’ to ‘uint8_t*’ {aka ‘unsigned char*’} [-fpermissive]
>   421 |         n = socketRecv(gXcpTl.MulticastSock, (char*) buffer, (uint16_t)sizeof(buffer));
> ```
> 
> ```
> /usr/bin/ld: CMakeFiles/xcpLite.dir/xcpLite.c.o: in function `XcpCommand':
> ~/xcpLite.c:1116: undefined reference to `XcpTlSetClusterId'
> /usr/bin/ld: CMakeFiles/xcpLite.dir/xcpLite.c.o: in function `XcpInit':
> ~/xcpLite.c:1349: undefined reference to `XcpTlSetClusterId'
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/xcpLite.dir/build.make:219: xcpLite] Error 1
> make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/xcpLite.dir/all] Error 2
> make: *** [Makefile:84: all] Error 2
> ```
> 
> Thank you!

You can use the patch from the pull-request.

mzurawski, 2021-12-11:
Similarly on the Raspberry PI4 I used cmake and compiled it   I made a few small modifications first due to the compilation errors.

xcpTl.h added: extern void XcpTlSetClusterId(uint16_t clusterId);
xcpAppl.c changed line 222:      *p = (char*) APP_NAME;
xcpTl.c changed line 421:         n = socketRecv(gXcpTl.MulticastSock, (char*)&buffer, (uint16_t)sizeof(buffer));

Afterwards I ran it up to about 40 MB/s on the DAQ rate with a few missing ODTs.

commit d77aed76 (2021-12-13): Fixes #10, minor clean ups, more flexible CMake build system (#11)

* Make CMake based setup more flexible

Don't pin compiler to g++. People can use clang too.
Don't pin build type to debug, let the user decide.

Describe CMake/Ninja based Linux and Windows build setup.

* clean compile

* clean compile Windows

* Cosmetics
closed 2021-12-13 by RainerZ