vectorgrp/XCPlite #12: XCPLite runs on the simulink model

Z-ZWBY, 2022-02-13:
Hello Rainer Zaiser
 
Thank you for yourwork and i have run the XPClite V4 on the ubuntu.
If i want to use my ecu model ，i should replace the ecu.c or ecupp.cpp, it is right?
And i think i can use the sfunction and TLC to compile your XCPLite in the simulink, and simulink model can use the XCPlite( to suppot the model working on the ROS and use the CANAPE  to calibrate and measure)

RainerZ, 2022-02-16:
Hello,

yes, a cyclic task like in ecu.c C_Demo (V5) would be the typical place to run code generated from a Simulink Model  and expose its parameters and signals via XCP. In this case, you have to implement a method to map XCP addresses in the Mathworks A2L file to the memory locations of your model input, output and parameters. The CANape Simulink Integration Package provides support to do this. For more information, please contact me directly.

Regards

Z-ZWBY, 2022-02-17:
hello Rainer

Thank you for your replies.
About the CANape Simulink Integration Package, is it the Vector MATLAB/Simulink MC Add-On 15.0.10 (64 bit)? i read the description  and ask the my salesperson, the vector XPC Simulink only supports the simulink model and doesn't support the C code generation。
i hope my model could be compiled to CPP code and run as a node on the ROS. So i want to use the XCPlite and try it.

closed 2022-07-08 by Z-ZWBY