vectorgrp/XCPlite #14: Can this code act as Master side

aifei7320, 2022-02-24:
As Title. the demo code is act as an slave, can this code read an A2L file, and read data from ecu?

Z-ZWBY, 2022-02-25:
i think you should use Canape to read data from the demo

aifei7320, 2022-02-25:
Thank you, i want to record only a few messages through xcp, i have read some protocol, it's not very difficult to achive that right?  A CTO command is enough?

Z-ZWBY, 2022-02-25:
The code is designed for computer（linux and win platform）.if your code is running on the ecu, you should not use this code. You can download XCPbasic on the Vector web.i only use the Canape to read data and not try the CTO.
I only run the XCPLIte on the linux ,i want to use it to send message and calibrate on the ROS ,i  am trying it.

aifei7320, 2022-02-25:
Thanks for your advice.

closed 2022-02-25 by aifei7320