vectorgrp/XCPlite #15: The determination of the XCP 1.3 DAQ start time failed error

mucahitarduc, 2022-02-28:
Hi,

I am trying to integrate this XCP implementation into an android application. I could successfully connect Canape to the application over ethernet. However when I try to start measurement. It gives me the error below. What does this error mean and what should I do to handle that error?

Type	Time	Message	
Command released	-0.462s	Start measurement
-	-0.456s	CCSW: Clock informations
-	0.100s	1 Recorder started with recording
error message	0.101s	CCSW: The determination of the XCP 1.3 DAQ start time failed
Warning	0.307s	Warning from recorder 'Recorder': No data recorded

Thanks

RainerZ, 2022-03-02:
Hello,
this is most probably a problem with GET_DAQ_CLOCK_MULTICAST. 
There is a known issue with multicast and TCP.
Please try with multicast turned off. There is no benefit of multicast. It is just CANape default setting.
Goto DeviceManager/Protocol/Eventlist/ExpertSetting and set TIME_CORRELATION_GETDAQCLOCK to extended response.
Regards

mucahitarduc, 2022-03-03:
Hi RainerZ,

Great, this fixes the problem.
Now I wonder how should I set the base address for the library. I mean, I have implemented this library as native C++ library for android. It generates a libnative.so library and java application calls this library to run the application. If you can help, I appreciate that.

Z-ZWBY, 2022-07-08:
Hello Mucahitarduc,

I have the same problem. And i opened my Canape 17.0 and can't find the TIME_CORRELATION_GETDAQCLOCK setting.
Could you tell me which Canape verion you use and how do you set it?
Or i should use the new XCPLite V5

RainerZ, 2022-08-08:
"The determination of the XCP 1.3 DAQ start time failed" apears, when CANape gets no response to GET_DAQ_CLOCK_MULTICAST. Either disable Multicast in CANape or enable it in XCPlite.
 XCPTL_MULTICAST and XCP_ENABLE_DAQ_CLOCK_MULTICAST have to defined, when Multicast is enabled in CANape.

closed 2022-09-13 by RainerZ