vectorgrp/XCPlite #21: Incorrect IP address in the A2L file(XCPLite V5)

Z-ZWBY, 2022-07-11:
Hi
If i use the XCPLite V5 demo, i can not get the correct local ip in the A2l file ,the ip is always 127.0.0.1,but i fill in correct ip in the CANAPE and can connect to the demo. So i think there is little problem with the A2L.c, XCPLite V4 does not have  the same issue.

And about the 'The determination of the XCP 1.3 DAQ start time failed error', i disabled the 'XCPTL_MULTICAST' in the 'xcptl_cfg.h' and 'XCP_ENABLE_DAQ_CLOCK_MULTICAST' in the 'xcp_cfg.h',but there is still problem here.And in the Canape17.0 , i don't find the  TIME_CORRELATION_GETDAQCLOCK in the path 'DeviceManager/Protocol/Eventlist/ExpertSetting', do i need the latest version 20.0?
or i can fix bug by myself(base on V5 BUGFIX), becasue i used the V4, and i can't start measurement by using V5, i pay more attention on the V4.

George117, 2022-07-11:
Hi @Z-ZWBY  have you tried updating this line: 
- C: https://github.com/vectorgrp/XCPlite/blob/334206e1502ad6a84b4e09910537a12f6f952e90/C_Demo/main_cfg.h#L33 ?
- CPP : https://github.com/vectorgrp/XCPlite/blob/334206e1502ad6a84b4e09910537a12f6f952e90/CPP_Demo/main_cfg.h#L33

Or, start with XCPLite with option "-bind x.x.x.x"

Z-ZWBY, 2022-07-11:
Hi@George117

Thanks for your replying. if you do not define the 'OPTION_SERVER_ADDR', the A2L.C can get your local ethernet IP and write in the A2L.V4 has this function ,i run the v4 , but there are problems with V5

George117, 2022-07-11:
I see your point:
In _A2lCreate_IF_DATA()_ there is indeed this functionality implemented: _socketGetLocalAddr()_
https://github.com/vectorgrp/XCPlite/blob/334206e1502ad6a84b4e09910537a12f6f952e90/src/A2L.c#L298-L305

I'll give it a go on my system and see how it looks.

George117, 2022-07-11:
Unfortunately I dont use the MULTICAST in my implementation, so I cannot speak to it.

Z-ZWBY, 2022-07-11:
@George117 
How do you close the MULTICAST.
I disabled the 'XCPTL_MULTICAST' in the 'xcptl_cfg.h' and 'XCP_ENABLE_DAQ_CLOCK_MULTICAST' in the 'xcp_cfg.h',do you have the same way?

George117, 2022-07-11:
Yeah, exactly the same way:
[image]
[image]

George117, 2022-07-11:
> I see your point: In _A2lCreate_IF_DATA()_ there is indeed this functionality implemented: _socketGetLocalAddr()_
> 
> https://github.com/vectorgrp/XCPlite/blob/334206e1502ad6a84b4e09910537a12f6f952e90/src/A2L.c#L298-L305
> 
> I'll give it a go on my system and see how it looks.

#### So, I;ve replicated your issue. It seems to be a bug in the implementation.
[image]

Either define  **XCP_ENABLE_DEBUG_PRINTS**

or  move that copy operation out of the define scope:
https://github.com/vectorgrp/XCPlite/blob/334206e1502ad6a84b4e09910537a12f6f952e90/src/platform.c#L502-L505

@RainerZ @Z-ZWBY   I'm not sure if this is the correct fix but it seems to fix the issue reported (as far as I understood it)

Z-ZWBY, 2022-07-12:
@George117 Thanks for your work.
I think you don't use the Canape as the XCP master,is it right?

George117, 2022-07-12:
Morning @Z-ZWBY , you are right, I'm not using Canape.

George117, 2022-07-13:
Morning @RainerZ ,
Thanks for the confirmation!

closed 2022-08-08 by RainerZ