vectorgrp/XCPlite #22: XCPsim: Invalid counter in XCP transport layer header

balagopa, 2022-07-13:
In CANape 14,  We could see the sine waveform. But there are error messages in “comments”.

error message	0.676s	XCPsim: Invalid counter in XCP transport layer header: expected 22563, received 22570.
Warning	0.676s	XCPsim: Received incomplete DAQ list number 0: Missing data from 1 ODT(s) (from a total of 6).

Z-ZWBY, 2022-07-14:
Hello@balagopa,
which version do you run?V5?
Does the Canpe14 have the configuration the TIME_CORRELATION_GETDAQCLOCK in the path 'DeviceManager/Protocol/Eventlist/ExpertSetting'？

balagopa, 2022-07-14:
CANape ver 14 and Protocol type XCP5(dlc+ctr REtransmit). In that path 'DeviceManager/Protocol/Eventlist/ExpertSetting' we didn't have configured TIME_CORRELATION_GETDAQCLOCK

RainerZ, 2022-07-14:
Hello,
I created a experimantal branch V5_CANape14 with modifications for compatibility to CANape 14.
So far the issues I found in C_Demo project were:
- A2L V1.71 is not supported -> added a quick hack downgrade to 1.70
- GET_DAQ_CLOCK_MULTICAST is mandatory -> turned it on
- Created a CANape14 project in ./CANape14/
Could not reproduce the transport layer counter issue.
Did not try CPP_Demo. It makes use of A2L 1.71.

Please note thar the TransportLayer counter is used to detect data loss when using UDP as transport layer.

balagopa, 2022-07-25:
I done this step, here after we able to see the Error 	XCPsim: Invalid counter in XCP transport layer header:

RainerZ, 2022-08-08:
The "Invalid Counter" usually error indicates lost UDP packets, which happens at the limits of performance.
But this should not be the reason, because the data rate in the C_Demo example is pretty low (300KByte/s) by default and CANape and XCPlite are be able to handle 100-600 MByte/s depending on PC and network performance.

Does your yellow dt measurement signal (this is the measured ecuTask() cycle time) show a constant 2ms cycle time?
Try to increase ecuPar.cycleTimeUs and see what happens.
Currently it seems to me, that the behaviour of the Windows Sleep() function changed with Windows 11.

balagopa, 2022-09-12:
While trying to start the measurement we able to see **ERROR 22h CRC out of range** and parallel in write comment window showing ** Realtime support synchronization: None.** Can you please help me out this

RainerZ, 2022-09-13:
Could you please provide a trace window XCP log, to see which XCP command returns this error and why.

RainerZ, 2022-09-13:
"Realtime support synchronization: None" just means, that CANape software timesyncronisation is disabled.
Usually, you would enable time synchronisation of XCP timestamps to PC clock in Tools/Options/Miscellaneous/DeviceDriver/ Synchronization Software

closed 2024-05-27 by RainerZ