vectorgrp/XCPlite #38: Help

aaroniza0, 2023-08-30:
Hello,

I'm a senior computer engineering student working on my senior design project at the moment, my project requires reading some hardware through ethernet and the hardware lists xcp on udp/ip as the communication protocol. 

My team has to use a raspberry pi, would this program achieve this and work on the pi? If so, how would I install it on my raspberry pi?

Also, would it be possible to use python to read the values being read from the hardware?

Thanks,

aaroniza0

RainerZ, 2023-08-31:
Hi aaroniza0,

I recommend to use the branch V6_WorkInProgress.
It is the upcoming newer version to be released soon.
Will be easier for us to support and we are interested in feedback.

Regards
Rainer

aaroniza0, 2023-09-01:
I’m looking at the c++ demo in the V6_WorkInProgress but having trouble figuring out how I would collect data collected by the xcp slave, any ideas to accomplish this? This project I’m working on just needs to pull data from an xcp slave so i don’t think it’s too complicated to do, I’ve just never done this before.

closed 2024-05-27 by RainerZ