vectorgrp/XCPlite #42: Can XCPlite and CANape be run on the same machine?

f-zl, 2023-11-02:
Hello, I want to learn XCP by running XCPlite and CANape on one PC and debugging. I have set up a CANape project using the generated C_Demo.a2l. Clicking "Test connection" in CANape gives the result "Communication established". But CANape also reports all bus monitors and devices are offline, and the Online icon is disabled.

Can XCPlite and CANape be run on the same machine? Or do I need to run XCPlite and CANape on different machines, and connect them using Vector hardware? Thanks.

taiit, 2023-11-04:
Don't need Vector devices or other machine, just run both XCPlite application and CANape on same machine.
You can build C_Demo, run it. Open CANape and then start measurement.

RainerZ, 2023-11-06:
Hello,

if you have problems to connect, make sure the CANape XCP device (device manager/protocol/settings)  is set to the correct IP address (localhost 127.0.0.1).
The A2L file generated by the demos specify the IP address of the first ethernet adapter found in your system, unless not explicilty set on the command line (C_Demo -bind 127.0.0.1).
If CANape is set to automatic upload of the A2L file, the initial IP address setting has to be correct, to make the A2L upload possible.

Regards
RainerZ

closed 2024-05-27 by RainerZ