vectorgrp/XCPlite #44: Possibility using virtual address

truecarfield, 2023-12-05:
Currently I'm trying to use virtual address (e.g. 0x0, 0x1, 0x2, ...) for measurement signals in A2L.
For example I have 3 signals out1, out2, out3 defined as following:

`/begin MEASUREMENT out1 "" FLOAT64_IEEE NO_COMPU_METHOD 0 0 -1E12 1E12 ECU_ADDRESS 3 ECU_ADDRESS_EXTENSION 0
 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA
/end MEASUREMENT
/begin MEASUREMENT out2 "" SLONG NO_COMPU_METHOD 0 0 -2147483648 2147483647 ECU_ADDRESS 4 ECU_ADDRESS_EXTENSION 0
 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA
/end MEASUREMENT
/begin MEASUREMENT out3 "" UWORD NO_COMPU_METHOD 0 0 0 65535 ECU_ADDRESS 5 ECU_ADDRESS_EXTENSION 0
 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA
/end MEASUREMENT`

In this case CANape just give me one ODT entry with 8 bytes, can I somehow modify CANape settings so that it can give me an ODT entry with 8+4+2=14 bytes?

OliverKitt, 2023-12-05:
Hello,

you have to change the following expert setting of your XCP device to “yes”:
***@***.***

Regards,
Oliver

OliverKitt, 2023-12-05:
> Hello Oliver,
>
> thanks for the quick reply, which expert settings you mean?
>
> best regards

Hello,

It seems that my screen shot has got lost.

Please navigate to the device (dialog device configuration).
Then in the hierarchy
<Device Name>
  + Protocol
       + DAQ List

Then, on the right side, open the tab “expert settings” and under “General”, switch
ODT_ENTRY_ADDRESS_OPT_DISABLED to yes.

Regards,
Oliver

closed 2024-02-22 by truecarfield