vectorgrp/XCPlite #54: XcpServerReceiveThread can't be terminated cleanly

truecarfield, 2024-08-23:
### How to reproduce it?

Init a XCP server on TCP a after terminating a XCP server on UDP using XcpEthServerShutdown() within the same process

### What false behavior do you observe?

XcpServerReceiveThread isn't really destroyed when XcpEthServerShutdown() called, so the XCP server doesn't really reinited with TCP

### What correct behavior do you expect?

XcpServerReceiveThread terminated cleanly when XcpEthServerShutdown() is called

RainerZ, 2024-08-31:
Thanks
Please check the latest commit on master, problem should be solved

truecarfield, 2024-09-06:
> Thanks Please check the latest commit on master, problem should be solved

Thanks for the update, plz take a look on [this PR](https://github.com/vectorgrp/XCPlite/pull/55)

closed 2024-10-02 by truecarfield