vectorgrp/XCPlite #57: INCA COMPATIBLE

arturhercoles63, 2025-04-10:
I'm trying to generate a custom script using the XCPlite demo as a base. I managed to make it work perfectly with Canape, but when adding A2L to ETAS INCA I encounter the error:

Parsed file doesn't contain a valid ASAP2 project description

It seems that it is related to the generation of the A2L file

Could you help me?

RainerZ, 2025-04-23:
Hello,
sorry for late response.
Could you provide the A2L file?
If this is private information, maybe just the header part, where the project description is ?

arturhercoles63, 2025-04-26:
Sure

`ASAP2_VERSION 1 71
/begin PROJECT XCPlite ""
/begin HEADER "" VERSION "1.0" /end HEADER
/begin MODULE XCPlite ""
/include "XCP_104.aml"

/begin MOD_COMMON ""
BYTE_ORDER MSB_LAST
ALIGNMENT_BYTE 1
ALIGNMENT_WORD 1
ALIGNMENT_LONG 1
ALIGNMENT_FLOAT16_IEEE 1
ALIGNMENT_FLOAT32_IEEE 1
ALIGNMENT_FLOAT64_IEEE 1
ALIGNMENT_INT64 1
/end MOD_COMMON

/begin RECORD_LAYOUT R_FLOAT64_IEEE FNC_VALUES 1 FLOAT64_IEEE ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_FLOAT64_IEEE "" FLOAT64_IEEE NO_COMPU_METHOD 0 0 -1E12 1E12 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_FLOAT64_IEEE "" VALUE R_FLOAT64_IEEE 0 NO_COMPU_METHOD -1E12 1E12 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_FLOAT32_IEEE FNC_VALUES 1 FLOAT32_IEEE ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_FLOAT32_IEEE "" FLOAT32_IEEE NO_COMPU_METHOD 0 0 -1E12 1E12 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_FLOAT32_IEEE "" VALUE R_FLOAT32_IEEE 0 NO_COMPU_METHOD -1E12 1E12 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_A_INT64 FNC_VALUES 1 A_INT64 ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_A_INT64 "" A_INT64 NO_COMPU_METHOD 0 0 -1E12 1E12 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_A_INT64 "" VALUE R_A_INT64 0 NO_COMPU_METHOD -1E12 1E12 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_SLONG FNC_VALUES 1 SLONG ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_SLONG "" SLONG NO_COMPU_METHOD 0 0 -2147483648 2147483647 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_SLONG "" VALUE R_SLONG 0 NO_COMPU_METHOD -2147483648 2147483647 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_SWORD FNC_VALUES 1 SWORD ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_SWORD "" SWORD NO_COMPU_METHOD 0 0 -32768 32767 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_SWORD "" VALUE R_SWORD 0 NO_COMPU_METHOD -32768 32767 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_SBYTE FNC_VALUES 1 SBYTE ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_SBYTE "" SBYTE NO_COMPU_METHOD 0 0 -128 127 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_SBYTE "" VALUE R_SBYTE 0 NO_COMPU_METHOD -128 127 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_UBYTE FNC_VALUES 1 UBYTE ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_UBYTE "" UBYTE NO_COMPU_METHOD 0 0 0 255 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_UBYTE "" VALUE R_UBYTE 0 NO_COMPU_METHOD 0 255 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_UWORD FNC_VALUES 1 UWORD ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_UWORD "" UWORD NO_COMPU_METHOD 0 0 0 65535 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_UWORD "" VALUE R_UWORD 0 NO_COMPU_METHOD 0 65535 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_ULONG FNC_VALUES 1 ULONG ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_ULONG "" ULONG NO_COMPU_METHOD 0 0 0 4294967295 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_ULONG "" VALUE R_ULONG 0 NO_COMPU_METHOD 0 4294967295 /end TYPEDEF_CHARACTERISTIC
/begin RECORD_LAYOUT R_A_UINT64 FNC_VALUES 1 A_UINT64 ROW_DIR DIRECT /end RECORD_LAYOUT
/begin TYPEDEF_MEASUREMENT M_A_UINT64 "" A_UINT64 NO_COMPU_METHOD 0 0 0 1E12 /end TYPEDEF_MEASUREMENT
/begin TYPEDEF_CHARACTERISTIC C_A_UINT64 "" VALUE R_A_UINT64 0 NO_COMPU_METHOD 0 1E12 /end TYPEDEF_CHARACTERISTIC

/begin CHARACTERISTIC ampl "Amplitude of sinus signal in V" VALUE 0x121E8 R_FLOAT64_IEEE 0 NO_COMPU_METHOD 0 800 PHYS_UNIT "V" SYMBOL_LINK "ampl" 0 /end CHARACTERISTIC
/begin CHARACTERISTIC period "Period of sinus signal in s" VALUE 0x121F0 R_FLOAT64_IEEE 0 NO_COMPU_METHOD 0 10 PHYS_UNIT "s" SYMBOL_LINK "period" 0 /end CHARACTERISTIC
/begin CHARACTERISTIC cycleTime "Cycle time of demo event loop in us" VALUE 0x121F8 R_ULONG 0 NO_COMPU_METHOD 0 1e+06 PHYS_UNIT "us" SYMBOL_LINK "cycleTime" 0 /end CHARACTERISTIC
/begin COMPU_METHOD channel1.Conversion "" LINEAR "%6.3" "V" COEFFS_LINEAR 1 0 /end COMPU_METHOD
/begin MEASUREMENT channel1 "Sinus demo signal" FLOAT64_IEEE channel1.Conversion 0 0 -1000000000000.0000 1000000000000.00000 ECU_ADDRESS 0x12258 PHYS_UNIT "V" READ_WRITE SYMBOL_LINK "channel1" 0 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA /end MEASUREMENT
/begin COMPU_METHOD counter.Conversion "" LINEAR "%6.3" "" COEFFS_LINEAR 1 0 /end COMPU_METHOD
/begin MEASUREMENT counter "Event counter" ULONG counter.Conversion 0 0 0.000000 4294967295.000000 ECU_ADDRESS 0x12260 READ_WRITE SYMBOL_LINK "counter" 0 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA /end MEASUREMENT
/begin COMPU_METHOD RndVar.Conversion "" LINEAR "%6.3" "unit" COEFFS_LINEAR 1 0 /end COMPU_METHOD
/begin MEASUREMENT RndVar "Meu Sinal" FLOAT64_IEEE RndVar.Conversion 0 0 -1000000000000.0000 1000000000000.00000 ECU_ADDRESS 0x12268 PHYS_UNIT "unit" READ_WRITE SYMBOL_LINK "RndVar" 0 /begin IF_DATA XCP /begin DAQ_EVENT FIXED_EVENT_LIST EVENT 0x0 /end DAQ_EVENT /end IF_DATA /end MEASUREMENT

/begin IF_DATA XCP
/begin PROTOCOL_LAYER
 0x0104 1000 2000 0 0 0 0 0 252 1468 BYTE_ORDER_MSB_LAST ADDRESS_GRANULARITY_BYTE
OPTIONAL_CMD GET_COMM_MODE_INFO
OPTIONAL_CMD GET_ID
OPTIONAL_CMD SET_MTA
OPTIONAL_CMD UPLOAD
OPTIONAL_CMD SHORT_UPLOAD
OPTIONAL_CMD DOWNLOAD
OPTIONAL_CMD SHORT_DOWNLOAD
[... 8 lines cut ...]
OPTIONAL_CMD WRITE_DAQ
OPTIONAL_CMD GET_DAQ_LIST_MODE
OPTIONAL_CMD SET_DAQ_LIST_MODE
OPTIONAL_CMD START_STOP_SYNCH
[... cut ...]
/end PROTOCOL_LAYER
/begin DAQ
DYNAMIC 0 1 0 OPTIMISATION_TYPE_DEFAULT ADDRESS_EXTENSION_FREE IDENTIFICATION_FIELD_TYPE_RELATIVE_BYTE GRANULARITY_ODT_ENTRY_SIZE_DAQ_BYTE 0xF8 OVERLOAD_INDICATION_PID
/begin TIMESTAMP_SUPPORTED
0x01 SIZE_DWORD UNIT_1US TIMESTAMP_FIXED
/end TIMESTAMP_SUPPORTED
/begin EVENT "mainLoop" "mainLoop" 0x0 DAQ 0xFF 0 0 0 CONSISTENCY EVENT /end EVENT
/end DAQ
/begin XCP_ON_UDP_IP
  0x0104 5555 ADDRESS "192.168.100.200"
/end XCP_ON_UDP_IP
/end IF_DATA

/end MODULE
/end PROJECT
`

RainerZ, 2025-05-13:
Hello,

it seems that INCA does not support TYPEDEF in A2Ls at all.

XCPlite generates predefined TYPEDEF_MEASUREMENTs and TYPEDEF_CHARACTERISTICs for basic types, even if you don't use typedefs in your application.
A simple solution might be, to remove the predefines.
In A2L.c comment out the 2 lines:

			fprintf(gA2lFile, "/begin TYPEDEF_MEASUREMENT M_%s \"\" %s NO_COMPU_METHOD 0 0 %s %s /end TYPEDEF_MEASUREMENT\n", t, t, getTypeMin(i), getTypeMax(i));
			fprintf(gA2lFile, "/begin TYPEDEF_CHARACTERISTIC C_%s \"\" VALUE R_%s 0 NO_COMPU_METHOD %s %s /end TYPEDEF_CHARACTERISTIC\n", t, t, getTypeMin(i), getTypeMax(i));

Hope this helps.

Regards

closed 2025-05-29 by arturarrudaribeiro